freeDSP/freeDSP-aurora #14: Cannot connect to aurora

gdimopoulos, 2020-03-27:
I'm trying to connect to Aurora for the first time. I successfully pushed the WiFi configuration and the Aurora connected to my network. When I try the "read all parameters" in auverdionControl it fails. Same thing with webOTA when trying with a browser.

dspverden, 2020-03-27:
You should provide more detailed infos. With this limited description nobody can help you.
because you have to be able to configure the WiFi I assume you have been able to connect to the access point. Thus, you should check first wether you can do the read all via the access point connection.
The most common problems with the WiFi in a local WiFi networks are issues with firewalls, wrong router settings, wrong IP configuration in auverdionControl. Many problems are caused by Windows or AntiVirus software.
Therefore, we recommend to update to firmware 2.0.1 and use the web application instead.

gdimopoulos, 2020-03-27:
Sorry, you are right. The Aurora firmware version is 1.0.1. No add-ons or plugins. I'm running auverdionControl 1.0.1 on MacOS Catalina.

I did enable the AP mode and tried to connect but with the same result.

dspverden, 2020-03-27:
I am still wondering what you did.
1) You said that you have been able to set the WiFi credentials and that it connects to your WiFi.
Can you verify that? Is it listed with a valid IP in our router?
2) Can you show a picture of your settings dialog? I think there you did a wrong configuration.
3) What is the exact error message of auverdionControl?
4) When trying to read parameters in AP mode, did you switch auverdionControl to AP mode?
5) If yes for 4) did you connect your mac to AP-freeDSP-aurora?

I strongly recommend to update to 2.0.1. Please refer to the documentation for how to do that.

gdimopoulos, 2020-03-28:
Problem solved, after a couple of reboots I was able to connect with the auverdionControl and push different configurations. I also managed to upgrade to 2.0.1 as you recommended. Now everything works fine.

Thanks a lot for your help.

closed 2020-03-28 by gdimopoulos