freeDSP/freeDSP-aurora #16: Loud noise when configuring DSP via webinterface

archi, 2020-06-08:
In case this is relevant, my DSP plugin is a modified 8channels: All channels have their maximum delay cut in half (to make some space on the device); but only channels 1&2 have their maximum number of PEQ bands increased from 10 to 15 - by editing the existing PEQ. Else, it's the same, no new blocks added, none removed.

Anyhow: When making changes to my DSP configuration in the web interface, e.g. adjusting the parameters for the low shelv, **my loudspeakers make a deep and loud noise**. Sounds a lot like a fully driven output signal, e.g. rather unhealthy. If the Aurora was connected directly to the power amplifier (e.g. as preamp or active crossover), I am pretty sure this would cause damage to [some of] my speakers.

I work around this by muting the preamp, but since I plan to ultimately replace the preamp with the Aurora, this isn't an option in the long run. Not sure if relevant, but I'm currently connecting my "non-balanced" preamp directly to the board (pin assignment as you instructed me, double checked).

My power supply should be pretty stable at 7.20V DC, even with 1.5A dummy load. So I wouldn't expect an intermittent brown out in the DAC to cause this. More likely: I suppose some internal reset in the DSP to be the culprit. So I'd suggest muting the DAC via I2C while the DSP is reconfigured, this could avoid this kind of issue. E.g.: Mute DAC, wait a few ms, reconfigure the DSP, wait a few ms, unmute the DAC.
Any active audio briefly stream drops out anyway, so adding a few ms of additional silence shouldn't make a real difference.

If this only affects some DSP plugins (e.g. due to my channels having a different processing time from the modified PEQ bands?), this could be toggled using the HTTP API (I think "enabled" would be a sane default?).

I did not check the ESP32 code, at least not in great detail. So I am not sure how easy/feasible this is, but didn't want to withhold this potential source of trouble from you ;-)

archi, 2020-06-09:
Ha, the source say the DAC is already muted on every HTTP/JSON call. Hm... if you have any idea what I can try to debug, let my now. I only have an old [HM 312 oscilloscope](https://i.ebayimg.com/images/g/y4YAAOSwAbBe1lWw/s-l1600.jpg), so that's a little bit limited; but I can do "auditive" tests with my 5$ sacrificial speaker.

(Also: Switching to German is perfectly alright with me ;-))

dspverden, 2020-06-09:
Hi,

from your description I read that you have modified the 8channels plugin. Does the unmodified plugin produce the noise, too? Or does only the modified one make that noise?
If that is the case you may have to double check that alle the addresses are changed correctly (read it from params.ini) and that the GUI elements of the HTML page point to the right filter block.

archi, 2020-06-12:
I played around a bit, and I think the PEQ is at fault. I have to use some rather big gains (up to 18dB), and those seem to cause artifacts on DSP reconfigure; at least the boards output produces some ugly DC (7s video from my scope, just toggling the bypass of some control: https://photos.app.goo.gl/sewJ2kZ3popBjks38)

By manually entering the HP+PEQ settings into the vanilla 8channels plugin I could reproduce it. As a "solution" I now added a delay(500) to `softUnmuteDAC`. Together with the other 500ms delay that's in front of virtually every call to `softMuteDAC`, every change now produces ~1s of silence, but I think that's better than a blown tweeter.

zas, 2020-07-22:
@archi: which code did you change and how to fix your issue? I'm curious about delay for unmute.

archi, 2020-07-22:
As said above, but I can commit it to my branch after work ;)

zas, 2020-07-23:
> As said above, but I can commit it to my branch after work ;)

Actually you could prepare a PR against the current main repo. I guess the delay is applied after the unmute command, so it would be in sync with what I have done for mute in https://github.com/freeDSP/freeDSP-aurora/pull/35

dspverden, 2020-11-07:
Fixed: 58badc2
Pull request #56 is the fix.

closed 2020-11-07 by dspverden