freeDSP/freeDSP-aurora #19: aurora.ino:handlePostInputJson: Invalid use of http code 404 in case of failed deserialization

zas, 2020-07-07:
https://github.com/freeDSP/freeDSP-aurora/blob/21276bfe22873ccbef2ed4d0ad9af257ecf8831c/SOURCES/WEBAPP/ESP32/aurora/aurora.ino#L1862

Imho, it would be more appropriate to use http error code 400 with a meaningful message, perhaps something like.

```
request->send( 400, "text/plain", err.c_str());
```

zas, 2020-07-07:
Same issue at https://github.com/freeDSP/freeDSP-aurora/blob/21276bfe22873ccbef2ed4d0ad9af257ecf8831c/SOURCES/WEBAPP/ESP32/aurora/aurora.ino#L1932
https://github.com/freeDSP/freeDSP-aurora/blob/21276bfe22873ccbef2ed4d0ad9af257ecf8831c/SOURCES/WEBAPP/ESP32/aurora/aurora.ino#L2037
https://github.com/freeDSP/freeDSP-aurora/blob/21276bfe22873ccbef2ed4d0ad9af257ecf8831c/SOURCES/WEBAPP/ESP32/aurora/aurora.ino#L2145

zas, 2020-07-08:
Related code doesn't exist anymore in develop branch, so closing.

dspverden, 2020-07-18:
Fixed here: f833d43

closed 2020-07-18 by dspverden