freeDSP/freeDSP-aurora #31: Documentation: User Manual EN 2.1.0: page 20, potentiometer, rotary encoder, etc.

zas, 2020-07-09:
Those sections describe external devices that can be connected, examples of working/tested devices should be given too, as well as exact type of device.

For example, for the Volume Potentiometer, linear or log, resistance range, and few references would help new users.
The same goes for rotary encoder, IR receiver, screens, etc...

Perhaps working references could be grouped in a table at the end of the documentation.

zas, 2020-07-09:
Same goes for Power Supply connection, it should be made clear which kind of power supply is expected (7V, single output + expected power)

dspverden, 2020-07-09:
Ahm, both requests are already done in user manual 2.1.0. Even with links to successful used modules.

zas, 2020-07-09:
> Ahm, both requests are already done in user manual 2.1.0. Even with links to successful used modules.

Yes, I based my intial comment on previous doc, then saw you added them, so I modified it, perhaps you read initial version before modification.

Currently, we have a description of each connector, then a section with examples, I think about following structure:

- device1 (e.g. potentiometer)
  - connector description
  - compatible device description
  - examples of devices known to work
- device2 (e.g rotary encoder)
  - etc.

mike811, 2020-07-17:
Hi,
Page 19 of current German 2.1.0 manual: "Wenn Du ein I2C-Display anschließen möchtest, schließe es bitte ebenfalls an diese Stiftleiste an."
Users may connect their display to Vin (7-12V) according to this.

dspverden, 2020-10-03:
Fixed the typo on page 19: 89b92949f60f382ca76765154c1b1950259f8080

closed 2020-10-03 by dspverden