freeDSP/freeDSP-aurora #49: Aurora

dannyAux, 2020-07-25:


archi, 2020-08-17:
Hi @dannyAux, this is an empty issue - would you mind closing it, or adding something to it?

Falls du dich mit GitHub nicht auskennst - immerhin scheint dein Account neu zu sein - ein Issue sollte man eigentlich nur für Fehler oder ähnliche Anfragen (z.B. neue Features oder Unklarheiten) aufmachen. Wenn du der freeDSP-aurora Entwicklung folgen willst, kannst du oben rechts "Watch" auswählen, dann bekommst du eine E-Mail für jede Änderung/neuen Issue/... Wenn du "Star" auswählst, ist das sowas wie ein Lesezeichen, aber du bekommst weniger E-Mail Spam ;-) Falls du ein abgebrühter Profi bist, entschuldige bitte dieses Mansplaining, ich möchte nur höflich&hilfreich sein :)

dannyAux, 2020-08-17:
Hallo,

Danke für deine Hilfe. Das war tatsächlich nur ein Versehen weil ich auf ne neue App umgestiegen bin. Kann man das ganze wieder schließen ? 

VG
Daniel

archi, 2020-08-17:
Kein Problem :) Wie das in irgendeiner App geht weiß ich leider nicht ;) Aber über die Webseite ( https://github.com/freeDSP/freeDSP-aurora/issues/49 ) kann man unterhalb des Kommentareingabefeldes auf "Close Issue" drücken.

closed 2020-08-18 by dannyAux