freeDSP/freeDSP-aurora #64: Import and Export of 4FIR-Plugin Presets does not work.

Finnitio, 2020-10-31:
I get an error when importing an exported preset. At first the upload is confirmed:
[image]
Afterwards I get a second popup which says it failed:
[image]

dspverden, 2020-11-04:
I can not verify this bug. Please, can you provide some more information? Which FW/GUI version are you running? Does it happen with latest commit from develop, too? Which browser are you using? Did you try it with the supported browsers Firefox or Safari?

Finnitio, 2020-11-05:
I tested with Chrome and Firefox, for me the issue persists. It is GUI 2.0.3 and FW 2.1.2. I have to check with the latest permits at the weekend.

dspverden, 2020-11-05:
Maybe that is causing it: Your are mixing GUI 2.0.3 with FW 2.1.2. So yes please update to latest commit from develop branch and make sure that both are 2.1.3 then.

Finnitio, 2020-11-07:
Dont know how that happened. Anyways I tried flashing the develop today and the firmwar update via webota worked but I couldn't update the web interface. After installing the 4FIR plugin i endet up with: 
[image]
Will try thew master later on, though i suppose i have to get back to flashing with the serial adapter since i can't acces the webota.

dspverden, 2020-11-07:
Connect via Access Point and type URL 192.168.5.1/fallback then you can reupload your plugin 
Or replace 192.168.5.1 by the IP that your router assigned if you want to do it via your network.

Finnitio, 2020-11-07:
Okay, that worked. I found out that I again, was able to flash FW and gui from different versions. Now I have FW an 2.1.2 an gui 2.1.3. I am sure I flashed both gui and fw from the same develop folder, don't know why they are different versions. I Would suggest to implement a version check when uploading the gui, and maybe a notification in the web interface if there are differences. Should I be creating an feature for that request on github?

Now the import works!, can be closed. Sorry for the inconvenience

closed 2020-11-07 by Finnitio