freeDSP/freeDSP-aurora #67: 4FIR Plugin Input Selection visual reset to Analog1

MarsianCRaute, 2020-11-02:
Input selection is stored, but after reopening browser (FF, cookies cleared) it displays Analog1 for all channels. All selection are still correct, but not displayed correctly.
[image]

dspverden, 2020-11-02:
Can you show me the version numbers at the right bottom?

MarsianCRaute, 2020-11-02:
[image]
Should be 10d50f0 ...

dspverden, 2020-11-02:
Did you recompile the firmware from the develop branch? Seems that your FW is outdated.

MarsianCRaute, 2020-11-02:
I'm a simple man - I download things, upload things and complain when stuff isn't working.
What needs to be recompiled? I thought using the WebApp folder from develop is enough.

dspverden, 2020-11-02:
recompile aurora.ino

MarsianCRaute, 2020-11-02:
ESPAsyncWebServer.h: No such file or directory
ufff

dspverden, 2020-11-02:
You did read the build instructions? Anyway, I have just compiled it for you.
[aurora.ino.esp32.bin.zip](https://github.com/freeDSP/freeDSP-aurora/files/5476699/aurora.ino.esp32.bin.zip)

MarsianCRaute, 2020-11-02:
Pls forgive a n00b...
Case closed.

closed 2020-11-02 by MarsianCRaute