freeDSP/freeDSP-aurora #70: [Develop] 4FIR Plugin, Switching presets does not change inputs.

Finnitio, 2020-11-07:
I mainly use preset A and B to switch between digital or manual inputs unfortunately it does not change the inputs. Not sure what else might not be changing.

MarsianCRaute, 2020-11-07:
Version?  I have not compiled fix #66 yet, but previus build works nicely.

Finnitio, 2020-11-07:
Latest develop build

dspverden, 2020-11-07:
Did you recompile?

Finnitio, 2020-11-07:
why should I? I use the \freeDSP-aurora-develop/SOURCES/WEBAPP/ESP32/aurora.ino.esp32.bin and \freeDSP-aurora-develop\SOURCES\WEBAPP\plugins\4FIRs for testing.

MarsianCRaute, 2020-11-07:
Please take a look at #67 and learn from my mistakes. 
https://github.com/freeDSP/freeDSP-aurora/blob/master/SOURCES/WEBAPP/ESP32/README.md

I guess we need compiled stuff - we are new to this stuff. Of course we should learn how to recompile code and understand why it needs to be done.

Finnitio, 2020-11-08:
alright, expected the develop to be precompiled beacause a) there are precompiled files b) the master is precompiled. Will compile that later, if I have everything set up for programming. @dspverden which IDE do you use for coding?

dspverden, 2020-11-09:
I am using VScode fro coding and Arduino IDE for compiling. Both on macOS.
The Arduino IDE does not export the latest binary to project folder by default. You have to do that manually. And you know, I am a stupid man. I usually forget it, because on GitHub we have the releases for binaries and pulling develop is for getting the source code.
Btw. I have now compiled everything so far into a pre-release, perhaps that helps you.

Finnitio, 2020-11-09:
couldn't find the prerelease to be honest. But if I am right if I compile with Arduino IDE an Export the binary I can upload that via webota?

MarsianCRaute, 2020-11-09:
https://github.com/freeDSP/freeDSP-aurora/releases/tag/v2.1.3-alpha.1
Jep, that should work too.

Finnitio, 2020-11-09:
Alrigt, didn't know there was a difference between branches and releases. Probably beacaus some use branches like relases. Anywas  I compiled it myself and uploaded it via webota. Since my aurora is now daily used in the living room the usb2serial adapter is inconvenient. So the volume gets updated when restarting as do the inputs when changing presets. Thank you. I will be available for testing if needed and will try to implement some features myself.

closed 2020-11-09 by Finnitio