freeDSP/freeDSP-aurora #72: 4FIR Plugin, 2.1.3-alpha.1: Preset switching broken

MarsianCRaute, 2020-11-09:
As written in title, switching preset with latest FW takes forever (5 min+, nothing happens). The "Switching Preset" message stays, after refreshing connection is lost. Reboot is required. 
Preset import works, takes about 20 sec.

dspverden, 2020-11-09:
Can you open the web console in Firefox and have a look wether there are any error messages?

MarsianCRaute, 2020-11-09:
`TypeError: NetworkError when attempting to fetch resource.`
Is displayed after 20sec.

dspverden, 2020-11-09:
Ah, seems to be an issue with a timeout on your network or something like this.

MarsianCRaute, 2020-11-09:
Mhm, strange. Router reboot did not help. Reuploading plugin via /fallback seems to fix it.
I guess you will get used to this constant flow of non-issues. But I really tested a lot before reporting ;)

closed 2020-11-09 by MarsianCRaute