freeDSP/freeDSP-aurora #73: Apple Remote IR commands

MikaRau, 2020-11-11:
It seems that Apple's commands depend where you live? Atleast here in Finland default arduino code didn't work. 
Here are commands after sniffing them for finnish(?) users:
Up:	 77E150F7
Down:	77E130F7
Left:	77E190F7
Right:	77E160F7
Middle:	77E1A0F7
Menu:	77E1C0F7
Play:	77E1FAF7

zas, 2021-12-02:
It should be solved by https://github.com/freeDSP/freeDSP-aurora/pull/97
Is it possible for you to test and confirm?

MikaRau, 2021-12-03:
> It should be solved by #97 Is it possible for you to test and confirm?

I will test it once I get my amp/dsp combo back up and running (couple of weeks till I get my parts). Your solution seems really interesting.

dspverden, 2021-12-13:
For Apple remotes it should be fixed by this commit: 67fe23e

closed 2021-12-13 by dspverden