freeDSP/freeDSP-aurora #75: Plop on changing SPDIF-Output

OldKamuffel, 2020-11-14:
Cascading 2 DSPs via SPDIF:
Changing SPDIF-Output from or to 'SPDIF IN' results in a 'plop' on the target DSP.

Changing between other output types seems ok.

Saw on:
Fw 2.1.3 (alpha 1)
Gui: 2.1.3
Plugin: 8 Channel(Target) and 7.1(Source)
Addon: Balanced Live
Windows, Chrome Browser

dspverden, 2020-11-21:
Please can you describe it a bit more? What I read is, that you change the source for SPDIF output on board 1 and once you select SPDIF-IN as source, the second board plops? Right?

OldKamuffel, 2020-11-21:
Yes, I have first a 7.1 DSP and a second front speaker DSP connected via optical SPDIF.
When all is running, for example set 'output 1 and 2' as source for SPDIF out on first board and than changing the source of SPDIF out to SPDIF it plops.

In German:
Ich hab die Kette USB => 7.1 DSP => SPDIF für die beiden Front Kanäle => 8 Channel SPDIF.
Alles läuft, im vorderen DSP ist SPDIF out auf 'Output 1 & 2' gestellt, im hinteren DSP die 8 Eingänge abwechselnd 'SPDIF 1', 'SPDIF 2'. Stelle ich nun am vorderen DSP den 'SPDIF out' Source auf 'SPDIF IN' Ploppt es. Beim zurück stellen auf 'Output 1 & 2' ebenfalls. Das ist mir versehentlich passiert, als ich als 'SPDIF OUT' Source USB 1&2 nutzen wollte. Der 'SPDIF IN' ist bei mir nicht belegt. Umschalten zwischen USB, Analog und Output als SPDIF Source macht bei mir keine Probleme.

dspverden, 2020-11-21:
According to your last two sentences I assume the plop is caused by the open SPDIF in on board 1. Thus, the SPIF receiver cannot lock on an incoming signal. Can you you check, if the plop is gone when you feed in an SPDIF signal? If so, we need a way to check that users cannot select an open SPDIF input as source.

OldKamuffel, 2020-11-22:
Ok,
using 2.1.3-alpha2 now, same base behaviour.
After applying a signal on the optical SPDIF input, the plop is not reproduceable anymore.
Even if I disconnect the optical wire the plop is not reproduceable until I reboot the DSP.

dspverden, 2021-01-03:
What do you mean with "until I reboot the DSP"? Do you still have plops with your setup?

OldKamuffel, 2021-01-06:
Mal auf Deutsch um Missverständnisse zu vermeiden:

Ich habe die aktuelle FW "2.1.3 Combined Bugfixes" aufgespielt.
 
1. Der DSP wird eingeschaltet
2. Es liegt kein SPDIF Signal an
3. Beim Umschalten des SPDIF Ausganges von oder zu einem SPDIF Eingang gibt es einen Plopp
4. Ein SPDIF Kabel (optisch) mit Signal wird eingesteckt
5. Kein Ploppen mehr beim Umschalten
6. Das SPDIF Kabel wird wieder entfernt
7. Immer noch kein Ploppen beim Umschalten
8. Der DSP wird ein und wieder ausgeschaltet. >> "until I reboot the DSP"
9. Das Ploppen beim Umschalten ist wieder vorhanden.

Wenn es bei der Analyse hilft kann ich versuchen das Plopp Geräusch aufzunehmen.

dspverden, 2021-01-06:
OK, dann ist das Problem, dass die PLL im SPDIF-Receiver des DSPs sich noch nicht einrasten konnte, weil kein Signal anliegt.  Ich studiere mal die Register des DSPs, ob das irgendwie testbar z.B. mit weil ein Bit nicht gesetzt ist und daher abfangbar ist.

Ok, the problem seems to be that the PLL of the internal SPDIF-Receiver of the DSP could not lock because there is no signal present. I will study the register description of the DSP wether this situation can be identified by a bit setting and can be avoided.

volpx, 2023-01-18:
Just looking at the schematic I can see that SPDIF IN and OUT (thus RX/TX) are swapped.