freeDSP/freeDSP-aurora #77: AKM Alternative

octave, 2020-12-09:
Hi guys,

With the recent news regarding the disastrous fire at AKM factory in Japan, it's currently impossible to source their converter and they encourage seeking alternative suppliers at the moment.

I was wondering if some of ESS converters could be use as an alternative?

Thank you!

octave, 2020-12-16:
Found this TI ADC too: https://www.ti.com/product/TLV320ADC6140#product-details

dspverden, 2021-01-03:
I am aware of this issue. Currently after two month it seems still unclear wether AKM ADCs/DACs of Aurora will be back in production soon. Thus, I will wait a little bit longer before making a decision. Redesign with ESS might be a good option, but redesign will take a lot of time and the ESS chips are more expensive then the AKMs.

octave, 2021-01-06:
Good to know you are aware of it! All the recent news are not very optimistic. Some say about a year before we can see them back in stock...

JayShoe, 2021-01-07:
Wouldn't a great alternative be to just have an Aurora Lite with no DAC or ADC? Just buffer a few FreeDSP headers and have a tightly integrated sub-module with converters.

Zadagu, 2021-01-08:
> Wouldn't a great alternative be to just have an Aurora Lite with no DAC or ADC? Just buffer a few FreeDSP headers and have a tightly integrated sub-module with converters.

I think this is a great idea. It would even allow to build a microphone preamplifier module for the Auroa.

JayShoe, 2021-01-08:
> > Wouldn't a great alternative be to just have an Aurora Lite with no DAC or ADC? Just buffer a few FreeDSP headers and have a tightly integrated sub-module with converters.
> 
> I think this is a great idea. It would even allow to build a microphone preamplifier module for the Auroa.

You can build a mic preamp module for the Aurora now... Not only are the I/O's pin headers (analog), there is also one FreeDSP connector (digital).

May I also suggest exposing some buffered SPI busses? If designed thoughtfully, one can envision a sub-module that attaches directly to right angle FreeDSP headers allowing the updated system to be the same size as it is now. Parallel FreeDSP + SPI headers would allow for pretty much any expansion module one can think of.

dspverden, 2021-01-14:
Hm, I don't like the idea of an Aurora Lite. This would require programming skills to adapt for different DACs and ADCs. But from my experiences so far there are not so many Aurora users how have these skills. Thus, it would only be useful for a very small group...

dspverden, 2021-01-14:
> You can build a mic preamp module for the Aurora now... Not only are the I/O's pin headers (analog), there is also one FreeDSP connector (digital).
>
> May I also suggest exposing some buffered SPI [...]

SPI? There is now SPI-bus on Aurora. And I don't see any free pin of the ESP32 to implement a SPI-bus.

kpek, 2021-09-28:
Hi, just wanted to follow up on this, any decision taken yet on probably replacing the AKM chips ? the prices are over the roof everywhere you look (~100$ per chip). 
It doesnt look like the production will resume any time soon (if ever) + not even trying to guess when will the retailers actually have the chips in stock so us mortals can buy them, since iam pretty sure theres huge backlog of orders already reserved for big players.
Thanks

MIcQo, 2022-01-02:
Hi, any update of this?

I want to build one, but this "unavailability" of AKM chips blocks me. I'm very unhappy from this situation.
i'm trying to find those chips few days but without success.

Thanks for answers.

JayShoe, 2022-01-11:
> Hm, I don't like the idea of an Aurora Lite. This would require programming skills to adapt for different DACs and ADCs. But from my experiences so far there are not so many Aurora users who have these skills. Thus, it would only be useful for a very small group...

Yes but then again when chips are no longer available, you aren't stuck redesigning the entire Aurora... Sure, have a "standard expansion module" with whatever chips you support but making it two boards (mainboard + module) means that the mainboard is designed once. It also means other users can expand on the Aurora mainboard.

djkylian, 2022-10-10:
It seems AKM is back in business, They started shipping out the first in Januari 2022 as I can believe this article:
[https://audioxpress.com/news/akm-resumes-sample-shipments-of-13-velvet-sound-dac-adc-products](url)

JohanAR, 2023-06-25:
Looks like Digikey has both AK4458 and AK5558 in stock now, so everything's good again, right?

Anyone have any inside info regarding Auverdion resuming production of these? Would be nice if one could buy a board with at least the smd components pre-soldered.