freeDSP/freeDSP-aurora #84: Parametric Equaliser import

artpelling, 2021-04-02:
As far as I understand there is no feature to import parametric EQ settings in the current web-interface. 
It would be great to provide this feature again (just like advertised in the youtube video).

Are there any plans for this? Or is there a workaround?

exislow, 2022-09-25:
This is also a major bugger for me. Would be great to be able to import PEQ settings. Is it also possible to have like 20+ PEQ settings per channel, e.g. in the 8channel plugin?

MarsianCRaute, 2022-09-26:
Maybe using _Import Preset_ with a small tool to create _Presets_  with predefined EQ settings is a "quick" solution. Not sure if there is a convinient way to import EQ settings into SigmaStudio?

As IIR EQ does not require lots of taps the number of PEQ's should have no practical limit. You'll need to create your own plugin in SigmaStudio though.