freeDSP/freeDSP-aurora #87: Is this project dead?

xtc0r, 2021-08-10:
I' checking the shop website for months to see when new plugins are available...but nothing...
Also I do not see any new commits or comments on issues on github. So is this project dead?
Would be a shame after all the money I invested in the hardware and case.

dspverden, 2021-08-12:
Well one way to keep such a project alive would be, that you provide some plugins. ;-)

Besides that, I am currently busy with making a new version without the AKM converters after the fire disaster at their factory. But that takes a lot of time. It is basically a complete new hardware development. Therefore, I don't have much time for new software features atm.

xtc0r, 2021-08-12:
Mit plugins meinte ich die Add-ons wie z.B. das Add-on B: [https://auverdion.de/produkt/addon-b-down-with-developers/](url) Das bräuchte ich um aurora so zu nutzen wie ich es möchte oder ist der Plan eher, dass jeder die  HW dafür selbst zusammen bauen soll und nicht aus dem shop bezieht?

Ich habe auch gesehen, dass du Code bei zas' FreeDSP fork hinzufügst aber hier nicht. Wird dort das Projekt weiter entwickelt oder spielt mir github da einen Streich? Evtl. Sehe ich das auch falsch.

dspverden, 2021-08-12:
Nein, ich füge keinen Code beim fork von zas hinzu. Hauptrepository bleibt dieses.
Die AddOns wird es auch dann wieder geben, wenn es wieder Aurora's gibt.

closed 2021-08-12 by dspverden