freeDSP/freeDSP-aurora #90: Visual Problem GUI, Input menu

MarsianCRaute, 2021-11-22:
Black line crossing through text. Independent from chosen input.
[image]

dspverden, 2021-11-22:
Which Browser is that? I don't have that issue here neither with Firefox nor with Safari (both on macOS).

MarsianCRaute, 2021-11-22:
FF  94.0.1, Win 10H1

dspverden, 2022-03-15:
HI,

please can you check wether the issue still exists with v2.2.2?
I can not recreate it here on my MACs.

MarsianCRaute, 2022-03-16:
Still the same. No big problem, I don't mind.

mike811, 2022-03-17:
Hi,

here it also does not look right:
[image]

xrtze, 2022-04-19:
Same problem, Win 11 64 bit. Equal behaviour in FF 99.0.1 and Brave 1.37.116
[image]

exislow, 2022-05-07:
I can confirm this on the latest macOS and latest Firefox.