freeDSP/freeDSP-aurora #104: No spotify or plex music over USB audio on windows 10

gdimopoulos, 2022-01-12:
I recently setup a new windows 10 PC and I tried to use Aurora with USB audio. All windows sounds play fine and I can stream audio from my browser (YouTube, Spotfiy, etc.). 

But when I try to play music from my Spotify desktop app it fails with an error ("Spotify can't play this right now"). I have the same problem when playing music with Plex.

In the device manager I can see the aurora-dsp devices recognized correctly but I also see an "auverdion DFU" device that is not recognized.

dspverden, 2022-01-13:
This is an issue of Spotify or of the UAC2 driver that comes with windows 10 because as you said playing via your browser does work. Aurora does not come with its own driver, it uses the driver from Microsoft. Therefore, nothing we can do (except somebody is willing to invest time for writing a driver for Aurora).

The auverdion DFU is related to firmware update of the XMOS chip. Windows does not support this while macOS an Linux do. Therefore, this is not related to audio playing.

Conclusion: This issue is not related to this project and therefore, I am closing it.

closed 2022-01-13 by dspverden

gdimopoulos, 2022-01-13:
From the documentation and the description of the project:
> Works with ASIO driver under Windows 10.
This means that this _is_ related to the project since you claim that Aurora works fine with the ASIO driver. If you have no plan of supporting windows users and fixing the related issues, you should remove this claim from the documentation and description.

gdimopoulos, 2022-01-13:
The list of multimedia apps that don't play audio with Aurora is not limited to Spotify. As I mentioned, the same applies to Plex and VLC when playing video with specific audio formats. And these are just the ones I tried. This implies that the issue is not with a specific app or player but with how Aurora receives and decodes the audio stream from these players.

I know first hand that there are devices that are capable of working with any player or content over USB without the need of special drivers. I don't know why Aurora can't do this but I know it is feasible.

dspverden, 2022-01-13:
I know what you mean, but again this is an issue for the Windows UAC2 driver and therefore, out of the scope.
It is proven that Aurora works class compliant and the apps you mention do play with Aurora on macOS which comes with a class compliant driver, too.
As I said there is no direct audio stream from those apps to Aurora. It all goes via the UAC2 driver that receives the stream from the app (perhaps decodes it) and then sends it in UAC2 format to Aurora. The fact that Aurora works with e.g. YouTube proofs that it can receive audio from the UAC2 driver and that the broken link is between the Windows UAC2 driver and Spotify etc. and therefore we have no influence on it.
As I said the way to circumvent it is:
a) Go via ASIO if your app supports it.
b) Write a new driver to replace the Microsoft driver (that is probably the reason, why other devices work).

And hey, this is open source here, make a fork and fix it if you think it should be fixed.

gdimopoulos, 2022-01-13:
Thanks for the detailed explanation. It does make more sense to me now.