freeDSP/freeDSP-aurora #105: DSP memory holds FIR coefficients in reverse order

mike811, 2022-01-16:
Hello,

tests reveal a maximum phase behaviour of current FIR implemetation.
This seems to be caused by incorrect order of coefficient upload. According to this:
https://wiki.analog.com/resources/tools-software/sigmastudio/toolbox/filters/firfilter
Coefficents need to be loaded in reverse order.

dspverden, 2022-03-15:
HI,

please can you verify wether the issue is solved in v2.2.2? I have added the coefficient order reversing to the ESP32 firmware.

mike811, 2022-03-15:
Hello,

seems to work correctly in principle with pre 2.2.2 but there is a difference in frequency response:
For example with 2.1.4 and manual coef reversal:
[image]

With pre 2.2.2 and without manual coef reversal:
[image]

Did something regarding PEQ change to cause this?

Regards
Michael

dspverden, 2022-03-15:
Are there PEQs involved in your setting?
Or do I see the FIR only there?
I did test it with only FIR active and directly from audio interface back into audio interface and saw the expected FIR response.

dspverden, 2022-03-15:
P.S. between 100 and 200 Hz I see a phase change, too. Can it be that you have enabled another filter (PEQ or Phase)?

mike811, 2022-03-15:
No Aurora PEQ were involved. All 10 PEQs had Gain 0. Tested with clean 2.2.1 and pre 2.2.2 FW. Only FIR-inside EQs generated via rePhase.
Didn't you mention something about extending PEQs to 10 in FIRs plugin?

dspverden, 2022-03-15:
Is there a way to show how the FIR should look like? So what did you design and does Aurora reproduce that correctly.
Or send me your filter and I can verify here.
Regarding PEQs: Yes the GUI was telling 10 PEQs but SigmaStudio has set it to 8 when copying the block from 8channels to 4FIRs.

mike811, 2022-03-15:
Yes, the first picture with 2.1.4 shows my Woofer with linear phase close to zero. This was only tuned by rePhase EQs and should look like that ;-)

dspverden, 2022-03-15:
ok, plz can you send me the text file you uploaded to aurora? I want to make sure that this is processed correctly.

mike811, 2022-03-15:
You got mail.

dspverden, 2022-03-15:
Yes saw it while I was typing here? Can I conclude from your previous post, that this issue is solved now?
Will have a look at your file anyway. Just to be sure with some files from outside of my lab.
Thanks for helping with debugging it.

MarsianCRaute, 2022-03-15:
Great! Can't wait to use FIR more often now

mike811, 2022-03-17:
Hello,
it seems there is still something going on with the FIR or EQ implementation.
After power up today I got this response measured:
[image]

Only after reloading the coef file again I got the nice response from two days ago...

dspverden, 2022-03-17:
I will verify this today evening.

dspverden, 2022-03-17:
Hm, I cannot reproduce that here. Could it be that you did not press "Store Preset"?
Because in the IR file you have sent me, I see that you equalize the dip at 100. So the first picture looks a bit like no filter present at all. Btw. the file you have sent me does not fully match your result because I cannot see the LP behaviour in your measurement.
This is the file you have sent me:
[image]
And this is what I measure after a reboot:
[image]

mike811, 2022-03-18:
Hi,
sorry for the confusion. I can't reproduce it here ether. You are probably right. Maybe I simply forgot to klick store preset when playing around last time...