freeDSP/freeDSP-aurora #108: v2.2.1 DSP creates 2 Acess Points

pillepalle1234, 2022-01-16:
One of my DSPs creates 2 Acess Points. Reinstalling the firmware and plugin did not solve this. It is not critical like my other issue with ignored PEQs but it is there...
[image]

dspverden, 2022-01-16:
Please, can you check wether other devices show the two access points, too? E.g. what does your phone see?

pillepalle1234, 2022-01-16:
Thanks for your answer!

The other DSP does not, but appearently your tip with using the phone is golden.
It only shows 1 access point.
So it seems to be a problem with the wifi config of my computer...

dspverden, 2022-01-22:
Ok, great. Then I will close this, because it seems not related to Aurora DSP.

closed 2022-01-22 by dspverden