freeDSP/freeDSP-aurora #116: stereoforever and The Room code artifacts

exislow, 2022-03-21:
hey @dspverden,

still a great job you are doing here. I have noticed that in the last commits code artifacts appeared, where you check against strings `stereoforever` and `The Room` to alter the behavior of the aurora board, for instance, see:

https://github.com/freeDSP/freeDSP-aurora/blob/5142f7cd48acabf1c630bbd7f1dddd0ec1198aa7/SOURCES/WEBAPP/ESP32/aurora/aurora.ino#L540

Is it possible, that this code handles only your private plugins? Have you committed it by accident?

dspverden, 2022-03-21:
No, the reason is a little bit different:
The plugin stereoforever will be released soon, it is in the final testing now, therefore, I decided to leave these checks in. Thus, I don't need a new release for the firmware for it.
The plugin "The Room" is a custom plugin that I made. In order to keep the firmware compatible I left the check there.
Regarding The Room please refer to www.hornsolutions.de and request the source codes there. The product is based on Aurora and falls under the license requirements of the CC license used by Aurora.

exislow, 2022-03-22:
Thank you very much for the fast clarification. I am excited about upcomming plugin.

closed 2022-03-22 by exislow