freeDSP/freeDSP-aurora #117: Proper code style: Refactoring

exislow, 2022-03-22:
Dear @dspverden,

I would like to contribute more and more to this project, since I really like it and use my aurora daily. I already have some custom modifications to this code held in my GitHub repository fork.

What I really have noticed is that the code styling is somehow odd. I couldn't figure out, yet, to what code style / IDE settings you stick to. I would like to propose to stick to the Arduino code style guidelines: 

* https://docs.arduino.cc/hacking/software/ArduinoStyleGuide
* https://blog.wokwi.com/how-to-write-clean-arduino-code/

If you agree, I could reformat the code of the `develop` branch and send you a pull request. What do you think?

This will make the code very readable and better maintanable compared to the current code style in my opinion (also other Arduino projects have shown sticking to a proper code style increases the coding efficiency).

dspverden, 2022-03-27:
Sounds good.
I would like to merge PR #97 first, if that is ok for you?
I also would like to talk about the style guideline first, because Arduino defines sometimes different styles for the same code.

exislow, 2022-04-02:
Sure, this is totally fine. Just tell me, as soon as the PR #97 is merged.

Of course, we can discuss the style guidelines first. Have you already had a look at the JetBrains CLion code style formatter? It sticks to what Arduino code style guidelines are recommending and also to a general propper C style. Tell me what you think about it.

dspverden, 2022-04-19:
I have merged PR #97 now.
I am now looking into style guide and what would be the best way to define it. Any idea? Something like a .md file, perhaps?

exislow, 2022-05-07:
An *.md sounds reasonable. You could also create a Wiki page here and refer to it. But I guess this is not a big deal at all.

Finding and committing to a proper style guide is the bigger task. Did you have a chance to check, e.g. the code formatter of CLion?

After PR #97 is already merged, I could also just send you a new pull request with properly reformated code so you would have a better idea what I am talking about. What do you think?

exislow, 2022-06-06:
*ping

exislow, 2022-09-01:
@dspverden are we still on the same page about this idea?

dspverden, 2022-09-01:
Just send. a PR and I will have a look.
I never worked with CLion or something like that. Lately I came across the Boost-style and I am now using that in all of my newer projects. But I see that this would be a lot of work to change aurora to Boost-style.

exislow, 2022-09-02:
Okay, I will do so based on the latest develop branch. CLion is just an IDE which used an general state of the art code formatter, which is absed on any specific code style. I will just send you a PR in the next days. A lot of files will contain changes, but just formatting changes, no logic changes.

dspverden, 2022-09-02:
ok, looking forward to it.

exislow, 2022-09-11:
I have got you a the PR #124. Let me know what you think about it :-)

closed 2023-10-17 by exislow