freeDSP/freeDSP-aurora #119: 8channel Plugin Gain and Phase Order

xrtze, 2022-04-28:
In the dsp.html of the 8channels plugin, the order of the gains and phases does not correspond to the channel-id. 
Is this intended?

xrtze, 2022-04-28:
Trying to build a custom plugin, I just can't make sense of the order, the phases and gains correspond to. Any hints?

xrtze, 2022-04-29:
Yes, i've read and followed the guide. I've also noticed the part in the sigma2aurora.py, which is correcting the ids. It does not seem to work though, because with the --gui option, the custom dsp.html is copied but the ids are not corrected. 
[6chpafir.zip](https://github.com/freeDSP/freeDSP-aurora/files/8589792/6chpafir.zip)

dspverden, 2022-04-29:
You can proof as follows:
Open the plugin.ini browse to the block with e.g. the gain block addresses.
After you have compiled you project in SigmaStudio (click on

dspverden, 2022-04-29:
Can you send me your source for dsp.html?

xrtze, 2022-04-29:
In the above archive, under plugins/dsp.html. Since the sigma2aurora does not modify anything, it is literally the same file.

dspverden, 2022-04-29:
No I mean the dsp.html that give with the --gui argument.
It should look similar to the file you find in SOURCES/WEBAPP/plugins/8channels
The file you provided specifies blocks that don't exist in the SigmaStudio file.

xrtze, 2022-04-29:
which blocks exactly?

dspverden, 2022-04-29:
eg delay block in channel 1.
If you give the file you have provided in the archive to the --gui argument of sigma2aurora it will fail, because it has the wrong syntax. Compare it with the file I mentioned above

xrtze, 2022-04-29:
Thanks for your help, rkv. 
and sorry for my stupid mistake! I did not realize, I was feeding sigma2aurora with the postprocessed dsp.html. now that i rebuilt everything based on the template + plugins from SOURCES/WEBAPP, everything works fine. 
I could just as well have beaten myself for 6 hours straight.

closed 2022-04-29 by xrtze