freeDSP/freeDSP-aurora #120: 7.1 Plugin: highpass instead of lowpass filter

pillepalle1234, 2022-07-17:
In the 7.1 plugin there is a lowpass instead of a highpass filter in the Subwoofer section.

While both filters might be useful, a highpass filter is needed in 99% of cases, even closed box subs profit from one 10hz and below when used for movies.

Most consumer AV-receivers already apply a lowpass filter to the LFE and the rerouted low frequencies from speakers are lowpass filtered already.

I would like to make my own plugin but the script does not work for me, see other issue...

Thanks.

dspverden, 2022-10-07:
I have send you a modified plugin already weeks ago. Does it work now?

pillepalle1234, 2022-10-07:
Hi,
yes it works flawlessly, thank you :)

dspverden, 2022-10-07:
Great, would you mind to share it with other Aurora users? E.g. open a GitHub repository for it?

pillepalle1234, 2022-10-17:
Here it is: https://github.com/pillepalle1234/FreeDSP-Aurora-7.1-Homecinema-Plugin-Highpass

dspverden, 2022-10-23:
Excellent, thanks!

closed 2022-10-23 by dspverden