freeDSP/freeDSP-aurora #121: Missing XMOS source files

fabianmuehlberger, 2022-07-25:
I don't know if it is intentional, but there is no main.c in the XMOS/src/core folder. Also, if the customdefines.h file is made by Ross Owen, XMOS Limited, please refer to the original source. A dedicated readme or Github wiki page would be helpful.

dspverden, 2022-07-25:
Not sure wether you read the manuals, but you need to download the USB Audio 2 framework from XMOS first and setup the environment as described in the manual.
The framework cannot be hosted in this repository due to legal stuff.

closed 2022-07-25 by dspverden

fabianmuehlberger, 2022-07-26:
I use XTC Tools for programming, XMOS seems to have stopped the development of its xTimeComposer, and I tend to use the newer tools they provide. According to the XTC Tools documentation, the MCU can be programmed with a "C-like" language using lib_xcore and they provide a minimal "hello-world" example written in C. That's the reason I asked and assumed there should be a main.c file to compile your code.

dspverden, 2022-07-26:
Well the main.c is part of the UAC2 framework. The aurora repository only contains the files the have to be added or have changes. As I said XMOS does not allow to host their framework. Therefore, you first have to download the framework, then add the aurora stuff to it.