freeDSP/freeDSP-aurora #125: TOSLINK (S/PDIF) does not work.

exislow, 2022-10-02:
I have an Apple TV 4 / HD which is connected to my old LG TV. My TV has an TOSLINK and Coaxial S/PDIF out. If I try to connect the TOSLINK or coaxial S/PDIF out with my Aurora and configure it accordingly no audio received / interpreted by the Aurora.

Maybe my TV is not compatible or whatever... So I have used an active HDMI audiosplitter, which has Cinch and TOSLINK out. So I have connected the TOSLINK out of the HDMI splitter to my Aurora. But once again: No audio is received / interpreted. 

If I connect the analogue Cinch audio out of the HDMI spliiter or the Cinch out of the LG TV the Aurora receives an audio signal successfully.

Any ideas, why I my Aurora cannot interpret any S/PDIF signals?  

Aurora FW: v2.2.2
I use these cables:
* [TOSLINK KabelDirekt – TOSLINK-Kabel, optisches Audiokabel – 1,5 m – Glasfaserkabel für Soundbar (TOSLINK auf TOSLINK, digitales S/PDIF-Kabel, Stereoanlage/Verstärker, Heimkino, Xbox One/PS3/PS4)](https://www.amazon.de/gp/product/B00FQK3S3A)
* [S/PDIF Goobay 50267 Koaxial Digital Audio Verbindungskabel, Cinch S/PDIF, Zweifach Geschirmt 75 ohm](https://www.amazon.de/gp/product/B000NDHK3E)

dspverden, 2022-10-03:
I guess you are using AddOn-B?
Are you sure that you selected the right input?
Did you try it with another source then the TV/HDMI-Splitter?
It could be that the TV outputs a wrong sample rate or something like that because there are users going into Aurora with SPDIF.

exislow, 2022-10-03:
Sorry, forgot to mention it: Yes, I do use Addon B.

This is how my input config looks like:
[image]

Is it correct?

I have no other sources besides the TV (TOSLINK & S/PDIG Coax) and the HDMI splitter. What would you suggest for testing? Are you probably aware of any HDMI splitter or any other device which corrects with the Aurora?

What sample rate is required by the aurora? In this case the TV and the HDMI splitter must have a wrong sample rate. This seems strange to me.

dspverden, 2022-10-03:
It looks correct.
I have tested the SPDIF input with a Focusrite Scarlett 18i20 and there it was working. I can test it again for you. But that may take a few days, this week is not so much free time left.

exislow, 2022-10-03:
Just to get it right:

1. `S/P-DIF Input`: This is my source.
  a. This will be mapped on `SPDIF L` and `SPDIF R` (next to the `HP` block), correct?
3. S/P-DIF Output: What is this actually exactly for?

You do not need to re-test it. I would try to get any other affordable TOSLINK source, which is able to work with the Aurora. Can you tell me, what sample rates or whatever boundaries I need to pay attention to?

exislow, 2022-10-08:
@dspverden: I would appreciate it, if you could give me a short answer.

In the meanwhile I have bought this configurable HDMI audio extractor: (HDMI 2.0 Audio Extractor 4K 60Hz PS5 1080P 120Hz 4:4:4 HDCP 2.2 18 Gbit/s D-olby Vision HDR DE-Embed SPDIF Optical 5.1CH 3.5mm Stereo L/R Audio Breakout Digital Audio EDID Management, Scaler 4k 1080P)[https://www.amazon.de/gp/product/B07TZRXKYG]

I have tried all possible settings on the HDMI audio extractor, but still no sound via TOSLINK. Should I be concerned regarding my TOSLINK ports on addon B or maybe there is something with the IC?

This is how my current Aurora config looks like ()for test purposes the midrange speakers should play the TOSLINK input): 
[image]

dspverden, 2022-10-08:
Ah you are right to remind me. Too many threads running at the same time... ;-)

Config looks right. I am going to test it here again. Just to make sure.
What should work in any case is a sample rate of 48kHz.
S/P-DIF-Output is the output on the aurora board. You can select which signal Aurora should forward to the output. Yet, the AddOn-B does not have an output, this is on AddOn-A.

dspverden, 2022-10-08:
P.S. Could you post a picture of your wiring? Perhaps a cable is missing or broken or something like that?

dspverden, 2022-10-12:
Sample rate 48k should work, at least.
You could check the MicroMatch-cable going from Aurora to AddOn-B. Sometimes these cables don't make a contact or the wire breaks.
Or you send me an email with your address and I will send you a new one. I think it is worth the try.

exislow, 2022-10-15:
I appreciate your help very much. Let me double check, if I have another cable with MicroMatch connectors somewhere. But the wiring of the MicroMatch connectors (Aurora to Addon B) is correct in terms of pin 1 of the Aurora is connected correctly with the other pin 1 of the Addon B?

exislow, 2022-10-16:
I have double checked the MicroMatch Cable X101 with my digital multimeter. Every pin has non broken wire from connector to connector, since I had no spare MicroMatch connectors anymore.  But still no sound from TOSLINK. But what I have seen is this:

[image]

Due to the PCB layout the X101 housing collides with the white right next to it. I really made sure, that there is still a working conenction from MicroMatch connector to the other MicroMatch connector. Also the very first pins are only relevant for S/PDIF (not TOSLINK) according to the scheme.

Do you think there is maybe an issue with something regarding the X102 connector cable? This controls the PCA9536D on the Addon B, correct?

Here are some recent pictures of my Aurora setup:

[image]
[image]

@dspverden: Can you suggest how to proceed? Should I get my oscilloscope out for debugging? Or do you have any other simplier ideas?

dspverden, 2022-10-23:
If you have an oscilloscope that would be helpful.
You can track the SPDIF-Signal from connector to DSP pin then. All measure points are easy to access. You can also check, wether the MUX on AddOn-B is switching correctly.
I see an extra PCB on X102. What is it for?

exislow, 2022-10-26:
It is just an X102 expansion board (see https://www.diy-hifi-forum.eu/forum/showthread.php?20465-Vorverst%C3%A4rker-m%C3%B6glich/page2&p=323042). But maybe it has a bug or whatever. I will try to remove it and do tests without it.

Tank you for clarifying, that there are no debug routines built it. I have an oscilloscope at home. I will just study the KiCAD layouts again and try to trace down the error.

exislow, 2022-11-07:
I was going to finally trace down what is wrong with my setup. Actually it has something to do with the X102 expansion board. Since I have removed it MUXing of the S/PDIF input is working finally. 

Sorry for all the hassle with this topic. Lessons learned: Do not load your base setup with a bunch of untested stuff. Do it step by step. Thank you very much for your help.

closed 2022-11-07 by exislow