freeDSP/freeDSP-aurora #126: Windows and enumeration idea.

djkylian, 2022-10-12:
I am wondering, is this xmos design routed the same as the Xmos multichannel explorer kit XK-AUDIO-216-MC-xxx.

Cause when i was experimenting with that kit I found an unorthodox solution to let it work as a multichannel audio card with enumaration in Windows. Which shouldnt give any problems for experimental use.
I know it works with Asio driver as well, but not for software that can only use the default windows sound card.

I found that the xmos chip is also used in proffesional audio codecs like Focus rite and Berhinger which look pretty much the same in design. 🤔 Both use a branded Theysicon driver 🤓 and have a 18 and 20 in the model number 🧐. Which made me think of using one of their PID, VID to see if i could get the explorer kit to work as one of those 💡.
This worked with the XU216 and also on an older XS1 processor on the test boards with the default xmos multichannel firmware 👊.

Maybe someone with a programmer can see if this will also work for the Aurora.

dspverden, 2022-10-12:
Hm, I don't know why you should do this hack?
Windows 10 & 11 come both with a UAC2 driver and support Aurora as default sound device. Only the channel names cannot be changed...
The only "issue" is: As far as I know the standard windows drive is actually the Thesyscon driver. Thesyscon offers two version one with high and one with low latency. The standard windows driver is the high latency version. :(
I am not 100% sure about this. Maybe I am wrong, at least that is what I could find out.

djkylian, 2022-10-12:
Cause the windows default and thesyscon driver only show as a stereo input and output soundcard and doesn't give the option to run as a multiple stereo input and output card. For DJ and DAW software it will be easier to route in and outputs.

Actually I am looking for a multichannel audio codec with only analog in and outputs and no extra features like all the potentiometer and padding settings on the behringer, but also without optical in and output, wifi, bluetooth.