freeDSP/freeDSP-aurora #127: S/PDIF Input Settings per Preset

exislow, 2022-10-23:
I have noticed, that these S/PDIF settings are not stored per preset but globally. 

[image]

This means, I can have only one saved setting, which will be used for all presets. Is there a reason for it? I would be very happy, if S/PDIF settings could also be stored per preset.

dspverden, 2022-10-23:
Because the other 50% of all users want it not to be stored in the presets. ;-)

exislow, 2022-10-23:
Haha got it. Is it maybe possible to have an additional option, which declares the S/PDIF globally stored or per preset?

dspverden, 2022-10-23:
Should be doable.

dspverden, 2022-11-04:
Oh, I shall make it?
I am planning the next release mid/end December. I think it can come with that release.

exislow, 2022-11-04:
If it just a "few minutes" work for you, I would really appreciate it. Otherwise I will try to have a look by myself.

dspverden, 2022-11-04:
I have put it on my ToDo-list now.

exislow, 2022-11-05:
Thank you very much, this is very kind of you.

Btw.. This totally does not make sense: https://github.com/freeDSP/freeDSP-aurora/issues/127#issuecomment-1288131869

If you could store the S/PDIF settings per preset, then somebody, who likes it, how it is implement currently simply sets the S/PDIF settings in the upcomming implementation in all presets to the same value. Hence, he has the same result as right now with the global setting: S/PDIF input / output is the same for all presets.

Thus, I wouldn't vote for another option which can decalre S/PDIF settings "global" or "per preset" instead I would save time on this and simply make the S/PDIF settings stored per preset, which can be configured to also have the exact same result as with the current implementation. What do you think?

dspverden, 2022-11-05:
A long time I was thinking like you. But unfortunately life isn't that easy. I did that what you suggest with an another parameter (I think it was the master volume) and got then tons of emails like: "Found a bug, master volume not working when switching preset, please fix it within 24h else my lawyer will send you a nice letter." ;)

exislow, 2022-11-05:
You are kidding me? Okay, unfortunately I am not much into these legal details but I am sure you will find a safe way. Sorry to hear, that you have made such negative encounters.

dspverden, 2022-11-05:
Well, I can tell you stories, you won't believe how some go crazy

exislow, 2022-11-19:
I guess this issue is obsolete, since this behavior is already implemented. I do not know what did not workout with my setup, while I was testing this initially. Sometimes the esp32 is answering the AJAX calls verrrrrryyyyyyyyyyy sloooooow, thus the page is stuck for some seconds with settings from the previously chosen preset. Maybe this was the case.

Since I have tested it again right now and must say, S/PDIF input is already stored per preset. Hence, no actions needed. Sorry for the hassle.

closed 2022-11-19 by exislow