freeDSP/freeDSP-aurora #130: Rklib.pretty location?

mellbergsimon, 2023-03-04:
I am working on an variation of this board, basically just changing the external connections to D-sub. Also found some issues ordering a PCBA from jlcpcb.

I were also wondering what KiCad version I need to work in to make the library connection work, I have tried 7.0 and 6.0.11.
Every footprint is in a library called Rklib. Is this the [freeDSP library](https://github.com/freeDSP/freeDSP-KiCadLibrary) that is 7 years old? Where can I find the footprints of all 

How do I get up and working without replacing basically every component. I am a new KiCad user also so sorry if this is something everyone should know.

Kind Regards.
Simon

mellbergsimon, 2023-03-04:
Have anyone of you managed to order the whole board assembled from jlcpcb for example? Is this possible with existing files?

xtc0r, 2023-05-10:
Have you managed to find the rklib? I have the same issue. I want to try to order a Addon B PCB at a PCB prototype manufacturer as it seems it's the only option to get one. The Arurora shop doesn't have any since years.

mellbergsimon, 2023-05-10:
> Have you managed to find the rklib? I have the same issue. I want to try to order a Addon B PCB at a PCB prototype manufacturer as it seems it's the only option to get one. The Arurora shop doesn't have any since years.

Hi, no I did not succeed, i started going through the components needed an if they were available on jlcpcb. There were way too many components either not in stock or EOL for me to continue. As I had a deadline.

I think because of this the design will require rework as the main chip itself is unavailable. I think a chip factory burned down or something.

Good information is to be found in this [link](https://www.diyaudio.com/community/threads/freedsp-aurora-dsp-with-8-i-os-usb-audio-s-p-dif-adat-bluetooth-and-wifi-contro.334055/page-32) 

So finally I decided to go another route and bought myself a xilica solaro qr1 for my use case. Obviously your use case would differ from mine.

I wish you good luck with your endeavour.

closed 2025-04-27 by mellbergsimon