freeDSP/freeDSP-aurora #133: compile fails

LumpiZupf, 2024-09-08:
Hey,

Does anyone still have a setup that compiles and cares to share the libraries and tools with version numbers?

This is what I tried:

Arduino IDE - 2.3.2
ESP32 - 2.0.4 / 2.0.5 / 3.0.4
ArduinoJSON - 6.13.0
AsyncTCP - 1.1.4
ESP Async WebServer - 3.0.6
IRemote - 4.4.0
U8g2 - 2.27.6

For Libraries I did stick to the version numbers from the readme, for the ESP Board Library I also tried some versions from the time when the repository was last updated. I get so many errors from different files that either I have missed something obvious or I need a specific ESP or IDE Version??

Cheers,
Christian.

Erik1803, 2024-09-13:
Hi Christian,
I've just tried compiling the WEBAPP/aurora.ino project using PlatformIO in VS Code and it seems to be working. I have no chance to flash it to an ESP right now, but I didn't get any errors. Used the latest version of all libraries.
Maybe my platformio.ini configuration is helpful:

[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
lib_deps = 
    https://github.com/bblanchon/ArduinoJson
    https://github.com/me-no-dev/AsyncTCP
    https://github.com/me-no-dev/ESPAsyncWebServer
    https://github.com/Arduino-IRremote/Arduino-IRremote
    https://github.com/olikraus/U8g2_Arduino

LumpiZupf, 2024-09-24:
Thanks Erik,

I finally managed to give this another go this morning. I did mess about with the libraries a bit but unfortunately, I already had them installed. Neither with the new or the recommended versions I could get it to compile...

The ticket was switching over to VS Code and PlatformIO!

I still get warnings, but at least it compiles. I might have to go back to the recommended versions on some of the libraries.

Unfortunately I have no time left today, but if I get it to compile the way it should and uploaded I will report back here.

Thanks again for your reply, since I am not familiar with the arduino world, I would never have cosidered switching to another IDE / compiler.

Cheers,
Christian.

LumpiZupf, 2024-10-13:
...I found a little bit of time this evening and the platform.ini below will compile the master as well as the development version without any issues or warnings. 

The me-no-dev libraries had some commits and one fix, but nothing major changed.
You might also use the official arduino remote library, which had lots of changes, but then you either have to use a 2.x version or convert to the new one.

Next time I have time I will try to program the board...

Cheers,
Christian.

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
    olikraus/U8g2 @ 2.27.6
    bblanchon/ArduinoJson @ 6.13.0
    https://github.com/me-no-dev/AsyncTCP
    https://github.com/me-no-dev/ESPAsyncWebServer
    https://github.com/dspverden/Arduino-IRremote

derwhalfisch, 2025-02-16:
Hi all

In VSCode + PlatformIO.
To get it to compile, I have had to go through web.cpp and change all of these lines to 'const' and make the integer argument a 'char' No idea yet if this has changed the behaviour!

from 
`AsyncWebParameter *idx = request->getParam(0);`

to 
`const AsyncWebParameter *idx = request->getParam("0");`

My platformio.ini:

```

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	olikraus/U8g2 @ 2.27.6
	https://github.com/me-no-dev/AsyncTCP
	https://github.com/me-no-dev/ESPAsyncWebServer
	https://github.com/dspverden/Arduino-IRremote
	bblanchon/ArduinoJson@^7.3.0`

```