freeDSP/freeDSP-aurora #134: Readme for custom plugin building missing?

Finnitio, 2024-12-01:
I just bought a second dsp board and now have the opportunity to continue building and testing a new plugin. At one point there was a a readme to build custom plugins. I can't seem to find it somewhere. If I would like to add new blocks like a mixing matrix, what files would I need to change to add this to the default template?

I know that it will not be easily done! But I am not completely aware off how the web front end, esp32 firmware and AUDAU/sigma firmware work together. What I mean by that is: if add a block for the web interface, that will set variables on the esp firmware which will change the paramters on the DSP chip, how is the logical chain of events?

Kind reagrds

Finn

derwhalfisch, 2025-02-09:
That readme is in the Wiki.

You can follow the process in vague terms through web.cpp & adau1452.cpp.

For example - see handlePostLshelvJson in web.cpp which is handling the 'POST' HTTP request for making a change to a low-shelf block. It seems to update a local map of all parameters held in RAM in the ESP32, then calls 'setLowShelving' in adau1452.cpp to mute,  write the updates to the appropriate registers in the ADAU1452, then unmute.

The ESP32 is aware of which register addresses are mappings for each block parameter because such a list is exported by SigmaStudio when compiling the plugin.

Finnitio, 2025-02-10:
Thanks a lot, that is great input and a good starting point I think.